sta-titansrobotics/27315-IntoTheDeep #5: Movement

J-Dawgggg, 2024-10-02:
Create movement with the mecanum wheels to be able to use all aspects and benefits of it.
Goals:
- Moving forwards, backwards, sideways (not turning. moving on the x axis of the bot/field)
- Moving Diagonal
- rotation
- field orientated movement.
- sensitive movement (if possible)

J-Dawgggg, 2024-10-02:
https://seamonsters-2605.github.io/archive/mecanum/
- mecanum wheel movement and guide help thingy

Ambrose-doc, 2024-10-03:
Use the website on top that allow us to code more smooth program for the robot, then also we can put some maschine oil in the weel( if allow)to  cause the weel can move smoother.

J-Dawgggg, 2024-10-03:
https://docs.revrobotics.com/duo-build/mecanum-drivetrain-v2/mecanum-wheel-setup-and-behavior
rev robotics guide on mecanum wheel movement

J-Dawgggg, 2024-10-04:
About the mecanum movement, I have discovered that I am able to move the robot with mecanum wheels:
- forward by rotating all wheels in the forward direction.
- backwards by rotating all wheels in the backwards direction
- left (without rotating the robot) by making the top left and bottom right wheels rotate backwards while the top right and bottom left wheels rotate forwards.
- right (the opposite of going left)
- diagonally to the right forwards by rotating the top left and bottom right forwards (you can move the bot diagonally left downwards by rotating the same wheels but backwards)
- diagonally to the left forwards by rotating the top right and bottom left forwards (diagonally right downwards can be achieved by rotating the same wheels backwards)
- rotating clockwise is achieved by rotating the left wheels forwards and the right wheels backwards (doing the opposite will rotate the robot counter-clockwise

closed 2024-10-07 by mrfabroa