sta-titansrobotics/27315-IntoTheDeep #10: Completing The Intake

WithoutShield, 2024-10-09:
By the end of the sprint, we should have our intake done.

Anand-nicrypto, 2024-10-09:
Day 1:

- Mateo wasn't here, so I couldn’t start actually building the model
- I did, however, gather the parts
- I also played around with the idea of a longer plastic sheet for the intake, to see if we were able to scoop it up quickly and transport without it falling out
- I used a mock plastic sheet (made out of cardboard) to test out the scooping mechanism, and we were somewhat successful

WithoutShield, 2024-10-09:
Day 2:

- We built the main structure for the intake system
- We couldn't cut the shaft so we built our intake after measuring 175mm so that it is the correct length
- We assembled the two pieces of vertical extrusion onto the horizontal extrusion and used pillow blocks at the end
- We used the pillow blocks to fasten the hex shaft (which has the mesh grabbers on it)
- We began working on the servo but we couldn't finish it. We decided to finish it on Day 3.

Anand-nicrypto, 2024-10-10:
Day 3:
Minor Setback:
- We assembled the servo only to realize we put it on the wrong way (into the bracket)
- We ended up having to re-assemble the bracket and the servo, which took a while (to screw in)
- We also realized that we couldn't test it because the servo needs to be plugged into a control hub 
- However, we came up with the solution to fasten a regular motor (from our old robot) and then just plug the battery in, so that we could test the build, without needing special equipment
- I also started to brainstorm what length we should cut the plastic, so that it is a little longer than the original (for the more efficient intake system we discussed before).

WithoutShield, 2024-10-11:
Day 4:

- We began the final steps of the intake
- First we had to cut out the correct shape from the corrugated plastic
- We used an exacto knife to cut the plastic but we used a pencil to make the holes (for screws)
- We used corner brackets to mount the cut corrugated plastic onto the horizontal extrusion
- After some contemplation we decided to extend the plastic longer than it needed to be so that it is easier to intake/hold samples

WithoutShield, 2024-10-11:
Day 5:

- We finished fastening the corrugated plastic
- We used a motor to test the intake
- It is able to pick up samples but in its current unconfigured state it "yeets" the samples backwards
- The intake is now complete

Here is a picture of the intake:

[image]

closed 2024-10-16 by mrfabroa