sta-titansrobotics/27315-IntoTheDeep #13: Sprint 2  - Build/Test/Revise

mrfabroa, 2024-10-16:


WithoutShield, 2024-10-16:
Anand and I are going to find a way to get the intake high enough to deposit a sample into the high basket.

- [ ] By the end of week one, we need to finish the basic assembly of the robot with the rest of the group to see what type of system we can incorporate to lift the intake
- [ ] Once we know what we are dealing with we will brainstorm ideas for the lift
- [ ] Throughout week two we will build and hopefully test our improved intake system to see if it can accomplish its tasks

J-Dawgggg, 2024-10-17:
I am going to try to get the assembled drive train moving and able to have the mobility that can be used for practicing purposes
- [ ] Figuring out how to get the direction of the left joystick as a value (apparently according to sources you need tangent and a formula)
- [ ] Incorporating the direction of the left joystick into motor speed so the robot moves precisely in the direction the joystick is facing (in relation to the direction the robot is facing)
- [ ] Figuring out how to get the magnitude of the left joystick to allow for speed sensitive driving in relation to how far/hard in one direction the joystick is pointed to.
- [ ] Programming the right joystick to control clockwise and counterclockwise rotations by moving the right joystick left (for counterclockwise) and right (for clockwise)
- [ ] Bonus: figuring out how to make the movement of the robot field-orientated (so the robot moves left/right/forward/backwards or any direction in relation to the field

Geckoing, 2024-10-17:
Now that the Forearm and claw assembly is complete the next steps are:

-  To attach the assembly to the robot 
- To code it to pickup and drop vertical and horizontal samples and specimens
- Help wherever I'm needed 
- Assuming all that is done I will practice my driving coordination with Anand to be competition ready

closed 2024-11-29 by Geckoing