sta-titansrobotics/27315-IntoTheDeep #14: Build/Test/Revise 3

mrfabroa, 2024-10-16:


J-Dawgggg, 2024-10-31:
Last week seemed to go well in theory, but it was discovered that the OpMode would not show up in the FTC Driver Hub. This sprint, Anand and I will continue coding the robot. He will focus on the arm movement motor and intake servos in the code. I will focus on getting the drive train fully coded with field-orientated movement.

Anand-nicrypto, 2024-10-31:
This week I am helping J-Dawg a little bit to code the robot. I will also figure out how to code the servo so that the intake will work properly with the triggers (Right trigger moves the intake forward, Left trigger moves it out). After we have finished the code for the intake, and tested to make sure all of the functions work properly, I will practice in game scenarios with Max (such as picking up samples, and dropping them) repetitivly until it becomes muscle memory. This will  ensure that we are prepared for competition.

WithoutShield, 2024-11-01:
This sprint I'm going to try to help with smaller sized parts of the robot's code and then send it off to Jadon so he can integrate it to the main TeleOp script.

closed 2024-11-29 by Geckoing