sta-titansrobotics/27315-IntoTheDeep #16: Four Bar Intake Implications

WithoutShield, 2024-10-16:
Here is a video I found on four bar:

https://www.youtube.com/watch?v=SuGy5VHyT_w

Anand-nicrypto, 2024-10-16:
Oct 16th:

- We came up with the idea to use a four-bar intake system
- However, we spent this period assisting other group members to help finish their projects (so we could see where to attach the 4 bar mechanism)
- We still need to figure out if we will have enough space the add the 4 bar intake system
- We will most likely begin the process of building it tomorrow, and if we discover that there isn't enough room later on, we can modify the design (use smaller bars, smaller gears, etc.)

WithoutShield, 2024-10-18:
Oct 17th - Oct 18th:

- We had a complete mental breakdown after realizing what we were getting into
- We began brainstorming with the help of Mr. Fabroa's whiteboard skills
- Anand wanted a double sided screw
- Jadon proposed to use the weird toothy gear belt thingy to push and pull the extrusion
- Now, we are deciding where to put it to allow it to fold and extend to allow us to reach the high basket while still meeting horizontal and vertical requirements

Anand-nicrypto, 2024-10-22:
Oct 22nd

- We FINALLY picked a proper design
- We are going to make a four-bar intake
- We attached it to the gear with the chain, so that when they moved up, the primary bars would also move up
- We attached gear onto the primary bars (that were attached onto the chain gears)
- We are currently speculating a way to get a motor onto a hex shaft so that the secondary bars can move
- We are planning to cut off the hex shaft from Max's other part so we can have one that is long enough
- After we have the hex shaft, all we have to do is attach a motor/servo and our intake system
- Then we will have a fully functioning four-bar intake system!

WithoutShield, 2024-10-24:
Oct 23rd

- Thanks to superb commucation, Anand and I were both able to visualize the same design and began the final stretch
- We found a hex shaft that was long enough so we don't need to cut anything
- We finished the first extension by connecting the extrusion to the gears on the claw assembly
- We then attached smaller gears to the end of that extension to allow the next set of extrusion to pivot and maintain the same elevation and angle for the intake throughout the extensions
- We made sure that the extensions did NOT go over the horizontal or vertical height limits
- We should be able to finalize the four bar tomorrow

Anand-nicrypto, 2024-10-24:
Oct 24th

- After further designing, we realized that if we were to add the second pair of extrusions AND the Intake (attached to the second pair), it would collide with the claw
- This meant that it would go over the width limit, as it would be pushed out, and was a major problem
- However, after thinking about it for a while and measuring, we realized that we didn't need a second pair of extrusions
- We realized that if we were to attach the intake's extrusions directly onto the gears of the first pair of extrusions, it would still be able to reach the high bucket
- We decided to do that instead, and properly secured it
- After that we were deciding on a way to add the motor to extend the intake 
- We realized that we could fasten mini extrusions onto the sides of the robot, and properly secure the motor
- We started fastening it, but hadn't the motor yet, so we will properly attach and wire the motors tomorrow
- Today was very productive

WithoutShield, 2024-10-27:
Oct 25th:

- Friday was mostly spent refining our previous build and double checking the horizontal and vertical height limits.

- The extra extension had a shaft that stuck out too far horizontally, so Ambrose and I took off the claw and arm assembly along with the intake to loosen the shaft collars and fix it. After reattaching, the horizontal limits were okay.

- The four bar is (for real this time) complete and we shouldn't need to adjust it anymore unless we find an error that needs to be changed.

- Anand and I will likely be assisting our other group mates next week.

Anand-nicrypto, 2024-10-28:
Successes:
- We FINALLY finished the four-bar
- The whole idea works (it can reach the samples and the high bucket)
- It also fits within the height limit, which is a relief
- We were able to attach the motor, so once it is programmed with code, it should work perfectly fine

Failures:
- We weren't able to extend it a little longer (because of height limitations), so it can barley reach the high bucket (we might have to "yeet" it)
- The first set of bars cannot move all the way down because of a pillow block blocking it
- This means that the robot will need to be pretty far back from the center zone in order to grab a sample

- Overall, we have completed our intended design in the intended time, so our project was a success

closed 2024-11-29 by Geckoing