sta-titansrobotics/27315-IntoTheDeep #23: Sprint 3 - Robot Drivetrain & Arm coding

J-Dawgggg, 2024-10-31:
Main Description in Sprint 3 - Build/Test/Revise Issue Page.
My Goals:
- [ ] Debugging the Drive Train because the .java code for the drive train originally did not show up in the FTC Driver Hub when selecting an Op Mode. Need to find out why that is and debug it so that the drive train is functioning as intended with field-orientated movement to have a robot moving as efficiently as possible across the field.

Anand-nicrypto, 2024-10-31:
My Goals

- [ ] Programming the intake to work using my controller
- [ ] Practicing scenarios with max

J-Dawgggg, 2024-10-31:
Monday - Tuesday:
Didn't know if the reason the code didn't show up is because the Control Hub was older and different. Was not able to test as the newer Control Hubs and Expansion Hubs were not available yet. Helped Ambrose with whatever he needed to fix on the robot arms.

J-Dawgggg, 2024-10-31:
Wednesday:
- Realized the code still did not work and Mr. Fabroa was not here so I could not ask him for the reason behind that.
- Decided to convert the code from java to blocks to see if the problem would be fixed and I would be able to run the OpMode and move the robot.
- Converted the code from java to blocks. However, the code did not work as intended. The code was supposed to get the direction of the left joystick using atan and then translate that value into motor power to make the robot go in the direction that the joystick is pointing to. Instead, it made the robot move forwards and backwards with the right stick x value and rotate clockwise and counterclockwise with the left stick y value. I changed the power code for the front right and back left wheels by reversing their rotation and that swapped the roles of the left stick y and right stick x so that the left stick y would make the robot move forwards and backwards. The right stick x would make the robot rotate clockwise or counterclockwise. The main problem here is that the robot is seemingly insensitive to the left stick x. The robot does not give the slightest recognition of input when the left joystick is moved left or right. It doesn't strafe left or right when given the input which is strange.

Anand-nicrypto, 2024-11-02:
Wednesday
- Now that we were all back, I showed my code to Jaydon and he had added a few things
- He made it so that when you move the joystick up, the arm would move up at a multiple of the amount (to more you move the joystick up, the faster it would go)
- He did this by multiplying the power by the position/degree the stick was pushed
- After that, Jaydon wanted to try fixing the drivetrain code, and me and Max decided to start practicing, to see if we knew what to do
- We spent the rest of the period practicing and were able to sort of get the hang of it, however, the intake still wasn't working

Anand-nicrypto, 2024-11-02:
Thursday
- This day was more of a filler day, where I worked around with everyone to see if they needed help
- I started by working with Jaydon a bit, but he seemed to be on top of things (I tried to help with the moving left and right all from one joystick, but we didn't get anywhere)
- After that, I helped Ambrose a bit with holding things and screwing stuff in
- I ended off the day by helping Mateo with the code for the servo so that once he got it to work, it would be able to function properly
- However, it turned out that he needed to turn the servo into motor code ("set power"), so some of my original code had to be replaced

Anand-nicrypto, 2024-11-02:
Friday
- I started the day by coming up with solutions for a problem we had for a long time
- If I were to move the intake too far up, it would flip behind, and be unable to return back
- First I was coming up with a solution to use a bar or something to stop it
- Then, Mateo suggested that we should implement it in the code, but that would be too finicky and wouldn't work properly
- Instead, I decided to preload a screw into the bar that holds the intake
- This makes it so that whenever the intake rises, the screw hits the pillow block every time it gets to the right point
- That took a while to make, as I had to nut both sides fo the screw, so that the screw would be perfect length
- After that was done, I decided to check in with everyone and see if they needed help, after testing the intake

Anand-nicrypto, 2024-11-02:
Successes and Failures
- Overall, this week was more of a break weak, where we were getting everything polished so that it would be ready for testing and competition
- I did achieve my goals of coding the intake to work on the controller and practicing a little with Max
- However, I do believe that we could have achieved more during this week, and the code problem held us back a bit (with ti not being able to move side to side)
- However, now that we are rested and have finished our polishing, we plan to make the most of next week

J-Dawgggg, 2024-11-04:
Thursday-Friday:
- I found a new code for the drivetrain entirely as the previous one was not working due to a variable bug. https://gm0.org/en/latest/docs/software/tutorials/mecanum-drive.html#field-centric-final-sample-code 
- the new code now uses field-orientated-movement based on the direction the USB ports are facing and also the direction where the logo is facing
- the code:
[image]
[image]
[image]
[image]
[image]

closed 2024-11-29 by Geckoing