sta-titansrobotics/27315-IntoTheDeep #28: Sprint 4: Programming

mrfabroa, 2024-11-05:
All members must create one or more of:

1. Functional TeleOp Program verifying the functionality of a subsystem.
2. Integration of all subsystem programs into a single teleOp program
3. Unique autonomous Program capable of scoring in autonomous period

For individual deliverable:

- Frequent upload of code to GH under TeamCode > BlocksBackup with comments on changes made. 
- Video of robot performing task

Anand-nicrypto, 2024-11-05:
I will be helping to fix the Teleop and ensure it runs properly. This means I will:
- Make sure all buttons perform the action we want (raising the four-bar)
- Try to add limitations in the code so that it does not over-extend
- Cut down on unnecessary blocks
- Polish everything so that it runs smoothly (making the four-bar sustain itself in the air when lifted)
-

J-Dawgggg, 2024-11-07:
I am working with Anand to ensure the TeleOp runs smoothly with all the requirements mentioned above AND a functioning, moving drivetrain that has great movement that fits to the specifications of our driver, Maximo Doewich Araya.

Along with the TeleOp, I will try my best to make an AutoOp for the right and left starting positions on the field. More information will be posted in Anand's Sprint 4 issue.

closed 2024-11-29 by Geckoing