sta-titansrobotics/27315-IntoTheDeep #29: Sprint 4: Completing The Team Portfolio

Geckoing, 2024-11-05:
This is the sub issue dedicated to the documentation of progress on the team portfolio.

Geckoing, 2024-11-05:
4/11/24 Monday:

Today was almost entirely dedicated to the lesson that Mr. Fabroa gave on the necessary tasks we will need to complete for sprint 4. This includes the team portfolio which I have taken lead on and will begin working on until my groupmates finish their necessary tasks and join me on completing this.

Geckoing, 2024-11-05:
5/11/24 Tuesday:

I created the canva document dedicated to creating the team portfolio digitally. This is shared with the entire group and Mr. Fabroa. I spent most of the class time combing through the https://ftc-resources.firstinspires.org/file/ftc/game/manual-06 to see what will be needed on the team portfolio. After that I began adding elements to the canva document which is the correct size of 8.5x11 inches. The link to the canva project is: https://www.canva.com/design/DAGVoagoxrM/YDRewJ4hZrwk8Sx3xSrwfw/edit .

Not much has been done but there is more to come.

[image]

Geckoing, 2024-11-07:
6/11/24 Wednesday:

Today we were told about the imminent arrival of our shirts and their designs. I spent most of the class troubleshooting how to change our team logo from the multi coloured greatness to a mono-tone single colour so it can be used on the team shirts. Then I  helped Anand and Jadon with experimenting with the driver code. Jadon managed to code the left stick to go forward in any direction it is pointed in, and the right stick spins the robot in place. Both of these are great achievements, however we ran into the problem of the robot randomly disconnecting.

Geckoing, 2024-11-08:
7/11/24 Thursday: 

Today I worked further on the portfolio; advancing to now have the cover page and half of the table of contents page done. I also assisted with taking photos for our presentation which Anand is currently working on. Some of these photos are on the cover page of the portfolio next to all of the required information specified by the manual (team number, name, motto etc).

[image]
[image]

Geckoing, 2024-11-11:
8/11/24 Friday:

I was at the Royal Winter Agricultural Fair with my business class and thus was not able to work on the robot or any related tasks today.

Geckoing, 2024-11-11:
11/11/24 Monday:

Today I studied previous teams portfolio to see what needs to be added. I specifically looked at how other teams formatted their mission statements, award section and meet the team page. This helped me plan out what we are going to include in our team portfolio. After lunch I helped troubleshoot the issue of our robot not being tall enough to ascend. I resolved the issue quickly as we just needed to adjust the brackets that hold the 1st arm assembly higher so the claw part can go above the low rung bar. I also helped Anand practice his control over the arms. 

[image]

Geckoing, 2024-11-12:
Refer to https://github.com/sta-titansrobotics/27315-IntoTheDeep/issues/30#issuecomment-2471359297 for progress done today.

Geckoing, 2024-11-14:
13/11/24 Wednesday:

Today we went to the board and performed a showcase. Anand and I got some good practice in for a solid 15 minutes before the robot started to break down again. We then spent the rest of the trip trying to fix the robot but it was unsuccessful due to the damages sustained.

Geckoing, 2024-11-19:
19/11/24 Monday: 

Today Mateo and I worked on the portfolio completing most of the "meet the team" page before lunch. We were slightly delayed as Mateo had trouble logging on and when Canva finally loaded it wiped part of the portfolio I had worked on (Still don't know why). After lunch Anand and I practiced driving the robot and scoring on both the low and high basket. We had varied success but the key factor is we are getting better.

[image]

Geckoing, 2024-11-20:
19/11/24 Tuesday:

Today was spent working on the portfolio and practicing driver coordination for the competition. Mateo and I were able to further progress on the portfolio completing the "Meet the Team" page and with Ambrose's additions our "about the robot" page is well underway. After that Anand and I continued practicing however we had to stop because of the deteriorating condition of our intake. The corrugated plastic of the intake was pretty much completely destroyed and needs replacing immediately which is what I started doing for the remainder of the class.

[image]
[image]

Geckoing, 2024-11-20:
20/11/24 Wednesday:

Today a lot of progress was made on the portfolio. Since Anand finished the presentation we were able to have 3 people work on the portfolio (Mateo, Anand and Me). Due to this we started and progressed through various pages of the portfolio including new building process and programming pages as well as the awards page. With a great effort tonight and on thursday we can easily finish the portfolio while making it an expecitional representation of our efforts so far.

J-Dawgggg, 2024-11-22:
Thursday the 21st:
- I added the process and problems we faced and overcame in our TeleOp Development. Mentioned how I needed to learn how mecanum drive works in order to code it. Did extensive research in the usage and translation of left joystick x and y inputs/values into radians so they can be used in the drivetrain motor power formula. Also mentioned how the right joystick x values are used to rotate the robot clockwise or counterclockwise.
- Anand and I added our process in developing the arm movement code. How it seemed simple at first, but was difficult due to the initial arm build with 3 motors and arm extensions. I mentioned how Anand was struggling to move the arm to the desired positions because of the many different inputs needed to be made, so I revamped the arm code where the buttons x,y,a,b can be programmed to move the arm motors to pre-set motor positions. this helped Anand a lot and works.
[image]
[image]
[image]

WithoutShield, 2024-11-22:
I spent most of today sitting with Jadon trying to figure out strategy before the competition. I made sure the starting positions worked for the auto ops we have done (other autos will be developed in hotel room) and that we were able to at least bring forward these two autos to this tournament. Then, I helped complete the portfolio as well as making sure our parts/hardware checklist was accounted for alongside Ambrose.

[image]

Geckoing, 2024-11-23:
22/11/24 Thursday:

Today I worked with the team on speedrunning the portfolio. With some time and effort after school we were able to completely finish it while including everything that was necessary and more. The final canva design can be found here: 
[Team Titans 4 Nuts Galore - Team Portfolio.zip](https://github.com/user-attachments/files/17877952/Team.Titans.4.Nuts.Galore.-.Team.Portfolio.zip)

Unfortunately some of the portfolio got cut off during the printing process meaning we had to physically write in some of the lettering. The damages were not that bad and we finished this as well with minimal issues.

closed 2024-11-29 by Geckoing