sta-titansrobotics/27315-IntoTheDeep #30: Sprint 4 Coding: Polishing/Fixing Teleop

Anand-nicrypto, 2024-11-05:


Anand-nicrypto, 2024-11-05:
Nov 5th
- Today me and Jaydon decided to test out our code only to realize that it doesn't work
- For some reason, the motor for the third part of the four-bar won't extend, so we had to fix it
- I tried to get Ambrose to reconfigure it since he must have messed it up before, so I helped him do that
- While he was doing that, I wrote the code to make it stop (break) if the power was set to zero on the motor
- We spent the rest of the class trying to figure out how to configure everything, but we were unsuccessful

Anand-nicrypto, 2024-11-06:
Nov 6th
- Today I did a bit of both, I helped Amrbosse to fix the code so that all parts were working, and fixed the wires and everything
- I also started to brainstorm ideas for the presentation, and decided to start it on Canva, so that we can design it nicely
- I am thinking of using our Nuts Galore logo on front, along with a team picture

J-Dawgggg, 2024-11-07:
Monday: Mr Fabroa spent majority of the class giving out important information regarding this Sprint and all the requirements in completing this Sprint properly and being ready for the tournament. Not much was accomplished.

J-Dawgggg, 2024-11-08:
Tuesday:
- Discovered that the issue of the "E2" Ultraplanetary motor and the "E3" corehex motor lay with the code an cables. Basically, for the code to operate the "E2" motor, it was set to gamepad1 instead of gamepad2. That was easily debugged and fixed. However, the "E2" and "E3" motors would not power when there was input. To fix this, we tested the code with new cables. The new cables proved to work and were promptly installed while the old ones were put aside.

J-Dawgggg, 2024-11-08:
Wednesday:
- The AutoOp mode for when the robot is starting in the right side of the field, closer to the observation zone than the teammate was created.
- For the moment, the AutoOp is capable of parking itself in the Observation Zone during the AutoOp Period to score some points. Further improvements will be necessary to maximize the amount of points that can be scored during the AutoOp Period. May hard code the robot to score a specimen or a high bucket sample and then park itself in the Observation Zone.
- At the end of the day, it was discovered that the current length of the arm cannot reach the Low Rung for Ascents. Need to move the arm further down the gear where it is mounted to reach the rung and score points for a Low Ascent or Level 2 Ascent.

J-Dawgggg, 2024-11-08:
Thursday:
- For some reason E3 corehex motor did not work. It seemed to be a cable issue again. I think we kept the same cable that was discovered to be broken on Tuesday instead of swapping it out. Will need to make sure it is swapped out.
- Was testing the drive train movement code and for some reason, the robot would seemingly "lag". Inputs would either be delayed and the robot would be stopped for a brief moment before moving again in the desired direction or the robot would stop  briefly even though there was already input being sent to the Control Hub before moving again. This was very weird as it had never happened before. The part of the code that was responsible for getting the joystick x and y values along with setting power to the motors were not changed. The only change made to the code was the addition of some telemetry to gather joystick values along with motor power values. The problem still has not been identified as there has not been enough time to test all the potential issues due to the other stuff needed to finish. Potential problems could be: power/battery problem, controller hardware problem with the inputs, wiring problem, driverhub problem, or a Control Hub problem (unlikely).
- Mateo and I tried to program our claw Servo that is responsible for picking up specimens and placing them on the specimen rungs, but the Servo did not seem to respond to our programming even after following the REV Servo Programming guide carefully. Will need to fix this and program the servo.

J-Dawgggg, 2024-11-08:
[Robotics Movement Basics.pdf](https://github.com/user-attachments/files/17682856/Robotics.Movement.Basics.pdf)

J-Dawgggg, 2024-11-09:
Friday:
- Added telemetry to "nuts" OpMode to get the joystick y data for the left and right joysticks. I also added telemetry to get the motor power on motors "E1" and "E2".
- Telemetry was added "NutsDropV2" to get the joystick x and y data for the left joystick and the x value for the right joystick. Added telemetry to get the power of the motors for the drivetrain.
- Tested the drivetrain movement on the field and it was field orientated, but midway, it turned into robot orientated movement. Was told by George that pressing either the "start" or "back" button on the gamepad would reset the forward facing direction of the robot based on the code. Will test further with Max the driver.
- Ambrose complained how the "E3" motor would not hold the intake arm in place and would flop down if power was not delivered to it even though there was "zeropowerbehaviour brake" enabled already. Told him to change the gear ratio on the motor and the intake arm to slow the drop of the intake arm. Also told him that if power was put into the motors, it would flap between two distances. There were only two options: either make the arm slow down in its droop or make it flap.
- Either Anand or I will have to bind different buttons to different tasks such as binding the "a" button on gamepad2 to raising the arm up to maximum height to deposit the SAMPLE in the High Basket for example.

WithoutShield, 2024-11-11:
Friday:

- Jadon and I tried to program the claw servo (we were unsuccessful)

- I programmed the AutoOp for the left position

- I didn't work on the presentation because Max wasn't here

Anand-nicrypto, 2024-11-11:
Nov 11th
- Today I decided to work on the code more than the presentation
- I started by brainstorming with Jaydon ideas for set positions that we could code the robot to go into (intake position, high basket position, and reset position)
- We came up with the idea to use the X, A, and B buttons to program those commands
- I was able to make a base code for each, making each arm move at separate times along with a sleep command
- However, the sleep command didn't work, and we were unable to figure out why
- We also ran into the problem that the arm would move up in separate stages, so I need to find a way to fix that
- Hopefully by tomorrow, I can fix the sleep function and make all the arms move simultaneously

Geckoing, 2024-11-12:
12/11/24 Tuesday:

Mateo and I worked on coding an auto op program to go from the Red alliance starting point to the edge of the field and move forward. Combining this with Anand's code to extend the robot arm it means we can now drive to the edge of the board move forward and fully extend in one swift program. The auto op will still need some fine tuning to be able to consistently get to the same spot every time but we will need to mark the positions of the samples on the field in order to fine tune.

[image]

Anand-nicrypto, 2024-11-13:
Nov 12th

- Today, I was able to properly code the x button and practice a bit of driving
- I realized that the problem with the x button was that the E1 and E3 were all backward (in terms of one of them needing negative power, and the other needing positive)
- Also, I had to incorporate gravity, which took a little while to figure out
- I realized I just needed -0.5 power to the E1, and 0.25 power to the E2, and then gravity would work its magic, bringing us from the base position to the high basket position
- I was able to code this test it, and get it to work, and I was able to do the groundwork for the A button, to put the robot in high basket position
- I also started practicing a bit with Max, and we realized we had a lot to work on (mostly timing wize)

J-Dawgggg, 2024-11-13:
Tuesday the 12th:
- I created a new TeleOp mode in java on the REV Client called realNutsDrop. Basically, I copy and pasted the code from NutsDropV2 and then translated the OpMode nut from blocks into java. That way, Max and Anand can practice moving the robot and the arms together.
- Tested the arm movements a lot today. There was this sort of gear-grinding sound at certain points. Still need to figure out why that is there. Ambrose suspected it may be the gears not being in contact enough. He moved the gear back into full contact with the other gear. Hopefully, it works better. Also, this issue might be the reason why the robot was unable to make a low-rung ascent. The power was not being transferred efficiently. Will need to test further.
- Max is pretty satisfied with the drivetrain movement at the moment. However, the robot seems to go from field-centric to robot-centric movement randomly, but is fixed when the robot/code is restarted. It is unknown why this problem occurs. Main suspicion is that the robot's field-centric part of the code malfunctions when the robot is receiving inputs for rotation and linear movement at the same time.

J-Dawgggg, 2024-11-14:
Wednesday the 13th:
- SCHOOL BOARD DAY
- Created the AutoOp mode sampleLeft. Still in the working process, but it is supposed to start on the left side and move to grab a sample in the submersible zone, depositing it in the high basket, then parking in the observation zone.
- A lot of testing was done today, but was cut short because of chain problems that needed to be fixed.

Anand-nicrypto, 2024-11-14:
Nov 14th

- Today me and Max started practicing, and we realized that we have a lot to accomplish
- Max doesn't really know how to drive with feild orientation
- And I kept breaking the robots back because of hyper extension
- Howeevr, we were a lot better than when we first practiced, and like anything else, practice makes perfect
- I developed a rythm of how to move the robot forwards to get into intake position, using proeper timing and the incorporation of gravitational properties
- However, after a little bit, our configuration stopped working for some reason, so that was a major setback that required a lot of time and attention

J-Dawgggg, 2024-11-17:
Thursday the 14th:
- Wasn't able to test the AutoOps sampleLeft and sampleRight. I am currently working on sampleLeft and plan to make the robot move laterally left, then forward, then rotate 90 degrees clockwise, lower the intake arm, move forward while the intake servo is active, stop the servo and the robot drivetrain, move the drive train backwards the same distance as it went forwards, retract the arm so it is in compact/default/folded position, rotate 90 degrees clockwise again, move forward a certain distance, rotate 45 degrees clockwise, move forwards a little bit, stopping the drivetrain and then extending the arms upwards for the high basket if possible (might do low basket if the high basket becomes too hard), retracting the arms into default position, rotating 135 degrees, then moving laterally right into the observation zone and parking there.

Anand-nicrypto, 2024-11-19:
Nov 18th:
- I started practicing more with Max and came up with a strategy on how to achieve a high bucket
- Instead of moving each arm up one at a time, I will move the first arm up, then tilt the second arm back and fully unravel the intake
- That way, when I lift up the second arm, the intake will softly hit the high bucket, which perfectly positions it up for dropping it
- We did this a couple of times as well as working on some of the code and stuff

J-Dawgggg, 2024-11-20:
Monday the 18th:
- Was working on the AutoOp sampleLeft when I was notified by Anand that the current controls for the robot's arm movements were difficult to control. He and Max were struggling to retrieve a SAMPLE and deposit it in the high basket, the biggest challenge.
- The problem with the current robot arm controls is the three motors "E1", "E2", and "E3" are all activated by three separate inputs. The E2 main arm motor is operated by the left joystick y. The E1 second arm motor is operated by the right joystick y. The E3 intake arm motor is operated by dpad up and down. It is extremely difficult for Anand to control the exact movements and timings for the inputs of the three motors and arms to score the SAMPLE in the high basket without potentially breaking parts and grinding gears. Anand planned to program the x,y,a,b buttons to set power to the specific arm motor for a specific amount of time in order to achieve various arm positions.
- George, from Team Titans 5, noticed Anand struggling and asked if we had coded the buttons. I told him Anand was in the process of programming them. He checked the code over and told us that we were able to set specific motor positions and then have our motors move to that position. I then asked George for help in doing so so that our arm movements can be more efficient and accurate. Will continue work on this Tuesday.

J-Dawgggg, 2024-11-20:
Tuesday the 19th:
- With some more help from George and some testing, I was able to program the b button on gamepad2 to make the arms extend upwards to reach for the high basket by setting power to the motors until they reach the desired position.
- This was done incredibly consistently and will be a major help in Anand and Max's coordination to score SAMPLES in the high basket for maximum points.
- Will ask Anand what other arm positions he wants pre-set to buttons x,y & a to aid in his robot arm control.
- Even though the arm code has been revamped with buttons to move the arms into pre-set positions, Anand is still able to manually move and adjust all the arms to fine tune the arms if they become a little flaccid after reaching the desired position with the buttons if the arm motors are not "busy".
- Will finish programming the buttons Wednesday and replace the current arm code with the new, revamped one so Anand and Max can practice their driving.
- Will also do testing with the sampleLeft AutoOp with the new position mechanic

J-Dawgggg, 2024-11-20:
Wednesday the 20th:
- Replaced the old arm movement code with the revamped one in the TeleOpMode realNutsDrop. Also, Max asked me to change the code to robot-centric movement instead of the field-orientated because the field-orientated movements works for around 20-30 seconds before it suddenly becomes robot-centric for some reason. So, I changed one if statement that was supposed to reset the IMU (the variable that made the drivetrain field-centric based on the direction of the USB ports and the logo of the control hub) when the "options" button was pressed to -> "if (true){" so that the imu is constantly reset and the drivetrain basically becomes robot-centric while having the code for field-centric drivetrain. A shortcut to save time in helping Max get the drivetrain setting he wants.
- I saved the old arm movement code in case by copy and pasting it into a new java OpMode and then downloading it.
- Added telemetry to get the inputted powers of the E1, E2 and E3 motors.
- Tested the OpMode realNutsDrop with Ambrose to see how the new changes are. The new changes to the robot arm code are a major upgrade. I was successfully able to grab a SAMPLE from the submersible zone and then use the button b on gamepad2 to power the motors and move them to a pre-set motor/arm location to extend the arms structure upwards and score the SAMPLE in the high basket. This process was done with relative ease considering the weight and length of our arms compared to the power of the motors. The revamped arm code allows Anand to extend the arms upwards to reach for the high basket without having to tediously input power into the motors for a certain amount of time, but it also allows Anand to adjust and give more power to the arm motors after the motors have reached their pre-set position in case they get weighed down.
- Still need to test the sampleLeft AutoOpMode and make adjustments accordingly. Will also probably help Mateo with his sampleRight AutoOpMode.
- Will also program the other buttons x,y, or a to a pre-set position for Anand's convenience.

closed 2024-11-29 by Geckoing