sta-titansrobotics/27315-IntoTheDeep #34: Sprint 5 (Week of Dec 2-6) - Fixing Robot Hardware Problems

J-Dawgggg, 2024-11-29:


Ambrose-doc, 2024-11-29:
@J-Dawgggg @Ambrose-doc will gofixing drivetrain

Anand-nicrypto, 2024-11-29:
This week me and Mateo will be working on the intake

This includes

- Fixing the front so that the flaps are the first thing that touches the sample
- Redsigning the top plastic part of the intake because it was preventing us form spitting out a sample
- We will most likely just redo the top part, and bend it properly
- We will also start to practice, to make sure our cahgnes actaull work and make it better

J-Dawgggg, 2024-11-29:
> * [ ]  Build Quality:
>   
>   * [ ]  Adding a 1.5mm spacer to the front right wheel so that it doesn't loosen.
>   * [ ]  Replacing gears that have evident damage on them with new gears
>   * [ ]  Moving the main arm gears to 1 inch above the tall extrusion to reach for the high basket and score more efficiently
>   * [ ]  Fixing our intake servo gear contact.
>   * [ ]  Having better wire management, ensuring wires are not unplugged through the movement of the robot mid-match
>   * [ ]  Putting a physical hard stop to prevent the arms of the robot from moving past vertical.
>   * [ ]  Adding the SPECIMEN claw back to be able to score specimens. Also customizing the SPECIMEN claw to grip the specimens better and easier.
> * [ ]  Redesigning the intake a little bit so that there is more area for the intake flaps to grip and grab onto the SAMPLE without moving before scooping it in, up and out where it can be scored or put in the OBSERVATION ZONE to be turned into a SPECIMEN
>   
>   * [ ]  More area for easier intake grip on SAMPLE
>   * [ ]  circular SAMPLE route. in through the bottom and out through the top
>   * [ ]  ensuring it doesn't get stuck in the intake. making sure it is ejectable

Geckoing, 2024-11-29:
open

J-Dawgggg, 2024-12-03:
- [ ] Monday goals: fixing the drivetrain wheel problems, ensuring all wheels have around the same tension, the standoff by the chains are replaced and a plastic covering is added over it to prevent damage to the metal standoff.
- [ ] Tuesday goals:
     - [ ] replacing the damaged small gears in the main arm gear system. adding it back and lifting the entire system up one inch above the vertical extrusion. 
     - [ ] adding the current specimen claw assembly design back to the forearm securely and properly according to instructions.
     - [ ] Anand, Max, and Mateo will start redesigning the intake. hopefully they move the bottom corrugated plastic further back to expose the intake flaps more and moving the servo gear and intake shaft gear together so they are contacted and won't grind.
- [ ] Wednesday goals:
     - [ ] if not connected tuesday, we connect the main arm frame/system back to the drivetrain.
     - [ ] reattaching the control hub and expansion hub to the robot
     - [ ] Can start on wiring of drivetrain motors and arm motors. 
     - [ ] Continue revising the intake to ensure samples picked up can enter through the bottom smoothly and exit around and out the top smoothly. Make sure samples are not stuck in the intake.
- [ ] Thursday goals:
     - [ ] finish the connection of all cables to the control/expansion hub in their proper spots
     - [ ] ensure the connection at the ports are secure
     - [ ] wrapping up the intake revision
friday goals are coming tmr