sta-titansrobotics/27315-IntoTheDeep #35: Sprint 5 - Intake Redesign

J-Dawgggg, 2024-11-29:
- [ ] Redesigning the intake a little bit so that there is more area for the intake flaps to grip and grab onto the SAMPLE without moving before scooping it in, up and out where it can be scored or put in the OBSERVATION ZONE to be turned into a SPECIMEN
     - [ ] More area for easier intake grip on SAMPLE
     - [ ] circular SAMPLE route. in through the bottom and out through the top
     - [ ] ensuring it doesn't get stuck in the intake. making sure it is ejectable

Anand-nicrypto, 2024-12-04:
Dec 3rd

- Today we were able to reassemble the entire intake
- We realized that the old ointake got super banged up somehow, so we desided to redisgn it
- We also moved it back a lot that way the flaps would touch the sample first, guarnteeing that we do not miss
- Me and Mateo were also deciding wether to put the top plastic back on or not
- We ultimatlet decided not to (becasue the plastic was so broken)
- However, if, during practice matches, we realize that we need the intake, then we will put it back on

Anand-nicrypto, 2024-12-04:
Dec 4th

- Since the robot has not been assembled yet, we couldn't properly test our intake, however, we were able to attach a motor to the shaft, to simulate the servo
- It did work much better than the old design, however, we ran into a problem
- Because we had the 90-degree brackets at the bottom, that would touch the ground first, leaving a little bit of a gap underneath the corrugated plastic
- This was a problem because when we tested, it would sometimes not be able to scoop up the sample because it would hit that gap (the plastic was too high)
- Because of this, we tried to reassemble the brackets but discovered that this would not work
- Instead, we decided to bend the corrugated plastic the right amount so that there would be less of a gap, and after further testing, we realized that this helped a decent amount
- As of now, we are finished the intake, so we are just waiting for the robot to be done so that we can test to make sure our solution actually works

Anand-nicrypto, 2024-12-05:
Dec 5th

- Today, Mateo was not here so I couldn't to much with the intake
- I was still waiting for Jadon and Ambrosse to be done with the first to arms, so I helped a bit with that
- I also was able to install manual stops for the inspection
- I took a metal 90 degree bracket and attacked it to the gears, that way when the arm gets to a certain point, it can't mvoe any further (becasue it is blocked by the metal)
- We also tested ot see if it could reach high basket, and it was able to reach with ease, os our robot is really starting to turn out nicely

Anand-nicrypto, 2024-12-11:
Dec 9th:

- While testing, we found that there was too much space at the top that the sample could slip out of 
- We added a makeshift stopper composed of a "column" and 2 corner brackets holding it up
- This was very unstable but it did work, and me and Max spent the rest of the period practicing with it

Anand-nicrypto, 2024-12-11:
Dec 10th:
- We are now officially able to get a sample into the high basket 🎉
- However, we were only able to get 3 before the time ran out, which was defiently less than we expected
- Also, the makeshift bar that we added kept falling off, or didn't work proprerly so we had to look into how to figure that out
- Regardless, me and Max continued to practice, and were able to get a decent ascension
 
The main problem with the rod was that if we moved it up and further, it would collide with the gears of the servo, so I came up with a plan to fix it:

- I will dissasmble the bar, and instead of moving it outwards (where it touches the servo), I will just stack it up, so that it doesnt hit the gears
- Then I will attach another column that goes out over flaps, that way it will keep the sample in, and allow the gear to mvoe freely 
- In essence, I am building the same column design as before, just one space higher
- I will enact this plan tommorow

closed 2024-12-12 by Anand-nicrypto

Geckoing, 2024-12-12:
9-10/12/24 Monday and Tuesday:

Now that we officially have the robot back in an operational capacity we can begin with our goal of practicing the tele-op phase. We timed ourselves across both days achieving a best result of 4 high basket samples + a level 2 ascension. This is good but we aim to have 2 specimens + 4 high basket samples by the end of a tele-op phase. While optimistic Anand and I are determined to make it happen and with further practice and coordination we should be able to complete it. Training Jadon to be a human player will also be needed as Ambrose cannot make it to Kingston and we cannot rely on our teammates providing a human player every match.