sta-titansrobotics/27315-IntoTheDeep #39: Programming and Refining AutoOps

WithoutShield, 2024-12-10:
For this sprint, Jadon and I will be working on getting our Auto situation up and running.

Referencing what Jadon wrote in the Sudbury Reflection Issue:

- [ ] Creating an AutoOp to score a SPECIMEN onto the HIGH CHAMBER then park in the OBSERVATION ZONE

- [ ] Creating an AutoOp to score 3-4 SPECIMENS onto the HIGH CHAMBER then park in the OBSERVATION ZONE

- [ ] Creating an AutoOp to score 4 SPECIMENS onto the HIGH CHAMBER, grab a SAMPLE from the SUBMERSIBLE ZONE under the chambers before parking in the OBSERVATION ZONE

- [ ] Creating an AutoOp to score 4 SPECIMENS onto the HIGH CHAMBER, grab a SAMPLE from the SUBMERSIBLE ZONE under the ASCENT ZONE before parking in the OBSERVATION ZONE

- [ ] Creating an AutoOp to score a SAMPLE in the high basket before performing a Level 1 ASCENT on the left ASCENT ZONE

- [ ] Creating an AutoOp to score 3-4 SAMPLES in the high basket before performing a Level 1 ASCENT

- [ ] Creating an AutoOp to score 4 SAMPLES in the high basket, grabbing a SAMPLE from the SUBMERSIBLE ZONE and then performing a Level 1 Ascent.
(prioritizing the specimen AutoOps)

We will divy up these autos and work on them in-class. We may also need to stay after school to program them as well.

We will do our SAMPLE autos for the left side starting position and our SPECIMEN autos for the right side starting position. This is because the HIGH BUCKET is closer to the left position while the OBSERVATION ZONE is closer to the right position.

WithoutShield, 2024-12-11:
Today, Jadon and I took a look at changing how our robot would move around during auto. In the past we set power to the motors and then cut it after a certain amount of time, which got the job done but wasn't entirely accurate. Jadon figured out that we could theoretically use motor positions for our movement. This functions by making the robot move forward (or in any other direction) and then stopping once a certain threshold is reached (determined by an equation that essentially calculates distance traveled). This would not only make us more accurate in auto, but would allow us to easily change the speed of the robot during certain sections of our script without having to change any additional values. We originally encountered a problem with this method where the robot would drive forward seemingly indefinitely, but while Jadon was away for a while I figured out the issue. The problem was the fact that our math would return a negative number (instead of positive) which would take a much longer time to reach the threshold than intended. I fixed this by taking the absolute of the entire equation, which made the script function properly. Jadon then coded brakes so that our robot could stop instantly on demand. Tomorrow will be spent on developing a simple specimen auto op with our new movement framework.

WithoutShield, 2024-12-12:
Today, we finished our first fully-functioning auto using the framework Jadon and I worked on yesterday. All it does is score a SPECIMEN into the HIGH CHAMBER and then park in the OBSERVATION ZONE. In total this should core us 23 points as Jadon mentioned the fact that the specimens on the chambers get you points when auto finishes and when teleop finishes. After school we begin work on a more complex auto which builds off of and expands from the auto we made today. Ideally it will score 3 total specimens on the high chamber, bringing our auto score to 63 points (big). Today was not perfect though. We encountered a few kinks in our programs that caused minor inconsistencies, some of which we still don't know how to fix. Tomorrow and likely throughout next week we will continue work on our autos, preferably staying after school to finish them ASAP to give us enough time to iron out the kinks and also grant more time to our drivers to run drills.

J-Dawgggg, 2024-12-15:
Monday Dec 9:
- We renamed the drivetrain motor names in the configuration from "1,2,3,4" to "FR, FL, BR, BL" respectively to allows us to identify which motor is which.
- I changed the hardware mapping for the realNutsDrop TeleOp to the new motor names from their old ones
- I tried running it, however, the back left motor did not respond to my inputs. It turns out that I hardware mapped the motor name "backLeftMotor" to the name "FL" instead of "BL". I changed that and it started to work again and respond to inputs correctly.
- I reprogrammed the specimen claw so that it opens to the central position when dpad left of gamepad2 is pressed and closed to right position when dpad right of gamepad2 is pressed.

J-Dawgggg, 2024-12-15:
Tuesday Dec 10:
- After a minute or so of testing, the back left motor stopped responding to inputs again. It had nothing to do with the code because it would not receive inputs at all from any other code, even if all the code did was set power to the motors. it was correctly mapped to the proper motor. I concluded that it was probably a cable connectivity issue. I checked the connection of the cable on the control hub, and it was not as tight as the other cables. I swapped it out and it started responding to inputs again.
- Also, the problem with picking up specimens with the specimen claw was resolved as a happy accident because the current distance of our claw to the floor is the perfect distance and angle to pick up a specimen from the floor if it is put in an incline. This solves the problem of not being able to grab specimens off the field wall. As the human player for the team, I will make sure the specimen is in the correct position for our drivers to grab it.

J-Dawgggg, 2024-12-15:
Friday Dec 13:
- As Mateo mentioned for Thursday's documentation, there were inconsistencies in our AutoOp where our main arm would not go down during some runs. Initially, the problem was that the code would stop after the robot moved forwards towards the chambers and nothing would be scored. So we put an if statement to determine that the main arm was not busy, set the arm to run without encoder, then lower the arm to score the specimen. I realized that the if statement could be the issue because sometimes, the robot would think that the main arm is still busy even though it is not visibly busy. I removed the if statement and just made the arm run without encoder and move. Ever since then, the main arm has always lowered and scored the specimen.
- For the complex specimen AutoOp, we have currently made the robot score a specimen and move 1 sample into the observation zone. We still need it to move 2 more samples into the observation zone before grabbing a specimen and clipping it on. We hope to clip on 3 specimens before parking. if it is possible, a fourth would be amazing.
- for improvements, we need to make our robot move faster in the sample retrieval area, but we cannot make the robot move at full power instantly as the sudden shift in weight throws the robot a little off balance. We are currently making the robot move slower at specific intervals at a certain speed before increasing the speed to a certain distance before going full power. I could make the robot go faster by reducing the "acceleration" time or by making a variable as the motor power that increases as the robot moves until it reaches maximum power. Further testing will need to be done to make the robot as efficient in its movement as possible to score the maximum amount of points in AutoOp.

J-Dawgggg, 2024-12-17:
Monday Dec 16:
- Today was a pretty rough day for us as there were quite a few problems that we found a few solutions for.
- The general problem revolved around the major inconsistencies with our AutoOp. Basically, the code would be the exact same in numerous trials, but for some unknown reason, the AutoOp would do something unexpected. For example, in one run, the robot would clip the held specimen, move back a bit, and then go super far right. Then in another run, it would clip the specimen, move back a bit, then go right and stop short of the desired stopping point. And lastly, it sometimes would altogether completely malfunction and move forward, slamming into the submersible and the chambers and attempt an ascent on the chambers with a specimen in its claws. These were some of the possibilities that could occur when running our AutoOp even though nothing in the code had changed.
- The, quite frankly, stupid solution that I discovered was that the robot would start running as we intended if we just restarted the robot on the driver hub before running the AutoOp. Yeah. that was the stupid solution that solved most of the problems we had today. I don't know what happened to our code and consistency because last week, it was golden. Our consistency with the code and Auto was amazing.  Hopefully with this solution, we can continue our complex specimen AutoOp development to score multiple specimens for a high amount of points.
- Another problem and crucial point to consider is the fact that there are, in theory, 5 specimens that can be scored during AutoOp. 1 being held by your robot in the start of the AutoOp period, 3 in the form of samples on the field that need to be delivered to the OBSERVATION ZONE, and 1 more by the human player at the start of the game. Our robot and drivetrain design is not as fast compared to other teams like Lo-Ellen Robotics or different teams that I have seen online. Our current plan of scoring specimens was the hang the held specimen, push the samples on the field into the observation zone one at a time, then score as many specimens as possible before the time runs out. the only problem with that is that the pushing specimens part takes a bit longer to do compared to using the intake, but is easier to program. That is the first aspect to consider moving forwards. Do we still want to push the samples or is it worth it to switch to the intake to save a bit of time?
- Another point to consider is that should we prioritize sample retrieval, or scoring the specimens. This all depends on the time it takes to retrieve all 3 of the samples. If we are able to score the same amount of specimens in our current progression/path compared to a path of scoring specimens as soon as they are available, then we don't need to switch. However, we may need to switch if we can score more specimens doing the method mentioned above where we score as soon as they are available, instead of waiting for all samples to be retrieved. Another option is a hybrid of the two options I mentioned in this point where we score the held specimen first, score the specimen held by the human player at the beginning of the match, then push the samples before scoring as many of those as possible with our setup.
- I will need to discuss these options with my teammates before moving forward with any one plan as it takes time to adjust.