sta-titansrobotics/27315-IntoTheDeep #40: Field Practice - Sprint 6: The Run Up To Kingston

Geckoing, 2024-12-11:
This is the official issue for tracking all drive team practice with the robot. From samples to specimens this is where all progress will be thoroughly documented with the goal of making a spreadsheet to track our scores and times.

Geckoing, 2024-12-11:
9-10/12/24 Monday and Tuesday:

Now that we officially have the robot back in an operational capacity we can begin with our goal of practicing the tele-op phase. We timed ourselves across both days achieving a best result of 4 high basket samples + a level 2 ascension. This is good but we aim to have 2 specimens + 4 high basket samples by the end of a tele-op phase. While optimistic Anand and I are determined to make it happen and with further practice and coordination we should be able to complete it. Training Jadon to be a human player will also be needed as Ambrose cannot make it to Kingston and we cannot rely on our teammates providing a human player every match.

Anand-nicrypto, 2024-12-12:
Dec 11th

- Today I fixed the intake a bit, and made it easier to spit out samples, however, I am not able to repair much because how the bar is already touching the servo plate, and can't be moved any further up
- However, the solution does work decently, so me and Max spent most of the period practicing
- We are now able to get 3 samples in the high basket during that time (we ran into a few complications involving wiring, so we didn't get to do a full proper test)

- Tomorrow we will let J-Dawg and Mateo work on their autos, and I might brainstorm a few ways to make sure that the sample can be spat out properly (maybe putting another bar in front)
- This will all be so that the sample doesn't get stuck inside of the intake, like during Sudbury

Geckoing, 2024-12-12:
11/12/24 Wednesday:

Today Anand and I practiced extensively while Mateo and Jadon worked on making the autos better. We encountered a few issues like the servo for the intake disconnecting a lot and then the cable for the intake servo even got caught in between the chain and tooth gear on the left side of our robot. Thankfully the cable is fine and with a few zip ties we won't encounter this issue again. Our best score was getting 3 samples into the high basket (because of the issues), but this is still pretty good considering we got an average of 4 samples in low basket per game in sudbury. I even stayed after school to try and help mateo troubleshoot the autos while Jadon was at band practice.

Geckoing, 2024-12-12:
12/12/24 Thursday:

Today Anand and I spent most of the first part of class repairing/modifying the intake, as to fix the problem where we can't spit out the samples since they get stuck at the back of the intake. We came up with the solution of making a bigger "back board" and moving adding another flap to prevent the samples from getting stuck on either side and also under the backboard. Using last weeks progress on the portfolio which was just refining certain sections and preparing to add more detailed sections for each award and our outreach attempts; I took the second part of class to help with whatever came up and to work on the portfolio.

Anand-nicrypto, 2024-12-13:
Dec 12th
- Today I mostly fixed the intake, we added extra bars in the top, like how I described before
- After practicing, we realized that the solution actually worked, and we are able to get high basket easily
- We practiced a bit more, but eventaully just let Mateo and Jaden practice.

Anand-nicrypto, 2024-12-14:
Dec 13th
- We were able to practice a bit, but were mostly helping out/watching Mateo and Jadon doing the auto
- We provided some feedback, and I helped to retreive the robot
- As of now, we are able to get 1 specimen and bring one back

Geckoing, 2024-12-16:
13/12/14 Friday:

Today we were able to get a decent time period to practice in which we even successfully hung from the low rungs achieving a level 2 ascent. During Mateo's and Jadon's auto practice they were able to get a specimen on the high rungs and push a sample back from the spawn to the observation zone. This would score us 20 points as is plus it allows us to quickly score a specimen when the game starts.

Geckoing, 2024-12-18:
16/12/24 Monday:

Add Text

Geckoing, 2024-12-18:
17/12/24 Tuesday:

Today was a great day for practicing as Anand and I were able to use the whole period. We practiced many things including submersible sample retrieval, full length drills and even ascents. Our best was getting 40 points in Tele-op and 20 in auto making 60 points total all by ourselves. If we couple that with an equal score from our teammate that gives us 120 points per match which should guarantee us several wins. With plans to stay after school on thursday we will be sure to improve on our averages and become even better.

Anand-nicrypto, 2024-12-18:
Dec 16th

- Jadon and Mateo were still working on the autos, so we accompanied them to make sure that everything was fine
- They were mostly just finetuning it because it was inconsistent
- We were able to practice a bit with our new intake design, and it worked well

Anand-nicrypto, 2024-12-18:
Dec 17th
- Today was a huge practice day for me and Max
- we struggled a bit with communication for the first bit until we decided to break it up into sections
- First, we practiced high-bucket and only high-bucket
- This allowed us to get good at positioning for dropping the sample, and for moving the arms back down (helped train our muscle memory)
- Then, we would practice getting samples from the submersible zone
- This was more just luck and a bit of timing, and we were able to come up with some ways to better our chances of retrieving a sample (through robot positioning and arm movements)
- Eventually, we put everything together and were able to score about 5-6 in the high bucket, as well as ascent
- We did a couple more "real game" scenarios with a timer, and were pretty consistent when it came to scoring
- We will practice more at Mateo's house

Geckoing, 2024-12-19:
18/12/24 Wednesday:

Yesterday Mateo substituted for Anand because Anand was sick. This led to Mateo and i practicing for the whole period and after school. Mateo was able to pick up the controls quickly and together we even equal our best performance of 40 points in tele-op and 20 points in autonomous.

Geckoing, 2025-01-06:
19-20/12/24 Thursday and Friday:

We practiced extensively in all areas of field performance. From scoring samples to practicing ascensions and hanging specimens we did it all. Mateo, Anand and I scored a new record of 82 points which is sure to win us a match when paired with a fully functional teammate.

Geckoing, 2025-01-06:
6/1/25 Monday:

Today was the first day back from break and I spent almost the entire period working on the assignment that was given to us (Computer Hardware and the FTC Robot). I managed to complete the entire first section where it asks to detail each part of a computer, its key specifications and then the FTC equivalent. I will use the first half of tomorrows period to finish this before moving on to either practice or helping with the continued work on our portfolio.

Anand-nicrypto, 2025-01-06:
6th

- Today we spent most of the class working on the assignment
- Jadon was able to work on the autos, while me, Max, and Mateo discussed the itinerary of the tournament and tried to hype each other up
- We have the mindset, now all we need is a bit more practice, so we will be practicing for the rest of the week

Geckoing, 2025-01-08:
7/1/25 Tuesday:

Today we spent the period practicing both driving and our autos. The autos are once again acting up for some reason but we seem to have it under control. Meanwhile practice is still going good. Anand and I have great coordination for picking up both specimens and samples. if the samples are scattered and numerous in the submersible then, we can retrieve those too. Anand encountered a minor issue where the main arm would get stuck on one of the drivetrain shaft collars if he extended the second arm whilst in the fully compact position. However after realizing we can't really fix that he became quite good at just working around it. We also did some work on the portfolio, adding a full page dedicated to our mentor Mr. Fabroa.

[image]